gasparka/spectrogram #11: What is the materiel for heatsink

Dastgsm, 2019-02-12:
where did you buy this material between the limesdr and the aluminum at the down. The form of this material is like gum. Where did you buy it ?

mariocannistra, 2019-02-12:
The technical name is thermal pad (short form of "thermally conductive pad").
You can just google "thermal pad" and will find several online shops offering them.

They are available with different thermal conductivity.  The higher the better since heat transfer would occur at higher speed.

I would avoid those too thin (0.5 mm) since I guess the bottom side of the Lime pcb could have something going through it and reaching the heatsink metal surface.

Dastgsm, 2019-02-12:
Thanks, and what is the more performant on cooling technic , a thermal pad or a  Thermal Adhesive Tape like on this link?
https://www.amazon.com/dp/B0751GYD6N/ref=sspa_dk_detail_2?psc=1&pd_rd_i=B0751GYD6N&pd_rd_w=aL16F&pf_rd_p=10ebaf99-73de-4f5d-a994-e7f5fc52f86f&pd_rd_wg=1PuFh&pf_rd_r=NB8QZJ95NHQ5SJTTXA51&pd_rd_r=dc1bf738-2e67-11e9-9785-9966c44d7ef0&fbclid=IwAR3Xjwk5TOE0kqOhjPrpgGx5uKnHJPyYeDp2T1bfxLTigxFCXTneBWrGnT0

mariocannistra, 2019-02-12:
I would say it depends on which you are comparing.
In the description of the amazon item of your link there's this sentence:

> "Thermal double-sided tape has a thermal conductivity of only **1.5 W/mk**, Can't provide good cooling results, If you need high thermal conductivity products, Please search (B074W99H6R), This is our highest thermal conductivity product, Its thermal conductivity is **6.0 W/mk**, It is the highest specification thermal pad on the market, but it is not very sticky..."

Instead, if you look at this thermal pad here https://www.amazon.it/Solutions-GP-Extreme-compontente-dissipatore-calore/dp/B01AAHM1WW/  you will see it has **12W/mK**  ... double of the thermal pad mentioned in that sentence (B074W99H6R).

Dastgsm, 2019-02-12:
should i need some glue if i use the thermalpad! or i just put it on the limesdr !?

Dastgsm, 2019-02-12:
should i need some specific glue if i use the thermalpad! or i just put it on the limesdr !?

gasparka, 2019-02-12:
Make sure you use something that is not conductive, i just ran into thermal paste that in fact was fully conductive.
Your linked product states 'Breakdown Voltage: 5KVAC' i.e. not conductive.

Dastgsm, 2019-02-12:
any link for buying a thermal paste that you use for it?

gasparka, 2019-02-12:
I did not use any thermal paste for this.

mariocannistra, 2019-02-12:
> should i need some specific glue if i use the thermalpad! or i just put it on the limesdr !?

my one was a slightly sticky thing...  I would not call it adhesive...
I just placed it carefully where needed and gently pushed things together:  pcb+pad+heatsink

i didn't do this on the lime board yet but on another project.  wil do the same in this case.

closed 2019-02-13 by gasparka