gasparka/spectrogram #13: Installation

Dastgsm, 2019-02-21:
Hey,
if i use a default gateware of limesdrmini on source for the installation! or i need some modification of it before use it! how to install proprely it?

Chears,

gasparka, 2019-02-21:
Gateware is programmed automatically when you execute the 'spectrogram' binary.

Dastgsm, 2019-03-31:
And how about the dimension of the heatsink with the thermal pad down to  the limesdrmini? and the dimension also of the heatsink with your fpga?

Chears,

gasparka, 2019-03-31:
These are just some random metals i found, i don't have the measurements.

Dastgsm, 2019-03-31:
Hey @gasparka , until now, about time , how long did you use it for a max? and how many times. I'm testing OAI with my limesdrmini and i break it because it's no heatsink! that why i would like more information for precaution!!!

gasparka, 2019-04-01:
I have used the version with large heatsink for hours and many times.

Dastgsm, 2019-04-16:
is it possible to give me the dimension just for having idea for the two heatsinks?

gasparka, 2019-04-16:
Sorry no, i don't have access to them anymore.

closed 2019-05-17 by gasparka