gasparka/spectrogram #14: [Question] - Is it currently possible to pass window coefficients to the FPGA gateware?

mariocannistra, 2019-02-21:
Hello.
I'm testing different IQ data processing approaches for radio astronomy, also using different windows before the FFT.

I see in the diagram [here](https://github.com/gasparka/spectrogram/blob/master/doc/lime_and_diagram.jpg) that a Hamming window of length 512 is used by the FPGA.

Is it possible to pass another window's coefficients?

In the past I've used different windows from [this paper](https://holometer.fnal.gov/GH_FFT.pdf) .
Some were implemented [in this code](https://github.com/nEDM-TUM/periodogram/blob/59d7eb5143358221ab60ae70f6482cc605aa7295/periodogram/windows.py) to which I added others specified in that paper.

I was thinking of generating those windows with 512 coeffs and passing to the FPGA ; is this actually feasible/possible (no FPGA experience) ?

thanks

gasparka, 2019-02-21:
Hi

Window coefficients are hard-coded in the FPGA code. They can be of-course changed, but this requires regeneration of the FPGA image.

Also note that values are quantized in the FPGA, i guess some of these windows may require very high precision, so this needs to be tested. Actually one cool/special feature of the Hamming window is that it works almost exactly the same even with only 8 bits - i guess this was one of the design goals back in the time.

Let me know if you have some special window in mind, i can test this and generate the FPGA image. If it is better i could replace the Hamming window with it.

mariocannistra, 2019-02-22:
Really kind of you, thanks.  I will run some comparisons to select a specific one for this test.  will update this thread with the selected window.
In the mean time another question: I've used "Quartus Prime 16.1 Standard Edition" some time ago to JTAG program a more recent version of the LimeMini GW (it was a pre-prod one, not upgradeable with the normal sw).   Is Quartus the only (and the right) piece of sw needed to regenerate the FPGA image? I was wondering if I could in the future "just update the coeffs file and regenerate" without touching anything else of the source folder...
Do you currently store the hamming coeffs in some sort of include file? I could probably programmatically write that file that starting from my python code...

gasparka, 2019-02-22:
Yes, Quartus is what you will need, i use version 18.0.
Window coefs are here: 
https://github.com/gasparka/LimeSDR-Mini_GW/blob/f18eba12596fac6de37e5b048c212f7addb2ac77/LimeSDR-Mini_lms7_trx/src/fft/top.vhd#L197

Basically the VHDL is converted from Python sources, that's why it is not so well structured. I might consider making a script that executes the full conversion flow and results in FPGA bitstream...