gasparka/spectrogram #16: 80Mhz

Dastgsm, 2019-05-17:
i have some question about the maximunband passant! why 80Mhz? and is possible also to transmission of 80Mhz of band passant data over limesdrmini or it's just for the spectrogram? simple curiosity!!!

Chears,

gasparka, 2019-05-17:
ADC samples with 160MHz, but transmitting this to FPGA will be trouble over current protocol. Also with 160MHz you would not get filtering.

This project focuses on the RX side, TX of 80MHz band should be possible.

Dastgsm, 2019-05-17:
in this link, https://www.crowdsupply.com/fairwaves/xtrx they said that it's 30Mhz !!! have you any idea or any link to do this rx transmitting 80Mhz? and why spec  limits it at 30Mhz, what the incovenient if it's more 30Mhz until 80Mhz?

gasparka, 2019-05-17:
Sorry i dont quite understand what you are asking for. Where did they write 30MHz?

Dastgsm, 2019-05-18:
on the link , i send you capture screen

Dastgsm, 2019-05-18:
i couldn't send the capture but at the table "Comparisons", search "Comparisons" on the page of the link and at the end of the table there are a scrollbar. scroll it at the right side and see at the top of table after this

Dastgsm, 2019-05-19:
Or have a look lso in this new link : https://www.passion-radio.fr/emetteur-sdr/limesdr-mini-667.html

gasparka, 2019-05-19:
I think Lime limited the bandwidth to 30MHz because of heat issues.

Dastgsm, 2019-05-20:
ok thank's so with 2 limesdr !!! we could achieve 160Mhz ! one limesdr-mini for modulation and one limesdr-mini for the filtering

closed 2019-06-30 by gasparka