gasparka/spectrogram #17: [Question] - Increasing the LimeSDR Mini's bandwidth to 80 MHz with GNU Radio?

0xCoto, 2019-06-29:
I'm a bit confused as to what the repository is about. Is it to share a software for the RPI to view a 80 MHz spectrum in real-time, or does it also highlight how 80 MHz of bandwidth is achieved, even though the LimeSDR Mini is officially rated at 30.72 MHz? How could I achieve 80 MHz bandwidth without using spectrogram (e.g. using GNU Radio)? Would I just increase the sample rate to 80MSPS and ensure the temperature of the Lime is constantly under 54 C, or is there more to it?

Thanks a lot!

gasparka, 2019-06-30:
>  Is it to share a software for the RPI to view a 80 MHz spectrum in real-time, or does it also highlight how 80 MHz of bandwidth is achieved, even though the LimeSDR Mini is officially rated at 30.72 MHz?

Both.

> How could I achieve 80 MHz bandwidth without using spectrogram (e.g. using GNU Radio)? 

IMO you cant - because you are throttled by the USB speed. This project does decimation in the FPGA - which makes data rate manageable.

> Would I just increase the sample rate to 80MSPS and ensure the temperature of the Lime is constantly under 54 C, or is there more to it?

According to my tests you can increase the sample rate to ~60MSPS without dropping, yes Lime overheats and may take damage. Also you may need to modify the LimeSuite to remove the 30.72 MHz limitation - i can help with this if you want to do this.

0xCoto, 2019-08-21:
Hi, thanks for your answers and sorry for my late reply,

Is it possible for GNU radio to tell the Lime to perform the decimation in the FPGA in order to reach 60 or 80 MHz? If not, could I re-program/control the FPGA using another program in conjunction with GNU Radio to achieve what I need?

gasparka, 2019-08-21:
Sorry i don't fully follow the question. You can get 60MHZ just by setting the sample rate, no need for decimation.
What exactly are you trying to achieve?

HB9DTZ, 2020-03-09:
Great work Gasparka ! 
Just modified two LimeSDR mini with heatsinks... and will try your promising solution.

gasparka, 2020-03-10:
I am glad that someone is using this :) Let me know if you encounter any troubles, maybe i can help.

tdbone1, 2020-10-19:
> Sorry i don't fully follow the question. You can get 60MHZ just by setting the sample rate, no need for decimation.
can we also set to 80mhz within gqrx on the amd64 laptop running ubuntu as long as we keep the limesdr-mini cool?

when i adjust gqrx over 50msps i dont see the usb3.0 bandwidth get more saturated.
the spectrum size increases but the actual data on the usb3 bus does not.

closed 2020-10-22 by gasparka