gasparka/spectrogram #18: CANNOT find device in docker, while it can be found in host.

juzhang74, 2019-09-07:
firefly@firefly:~$ sudo spectrogram --server_only
2019-09-06 10:16:58.449 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
2019-09-06 10:16:58.634 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
Digest: sha256:8b85e91c4605ae1cd9ae7fcbd3b1de8ff8a25bdb2fffc8d9eaf0ef4eff4a2f2a
Status: Image is up to date for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2019-09-06 10:17:02.962 | INFO     | __main__:probe_devices:83 - Probing for LimeSDR-Mini devices...
2019-09-06 10:17:05.486 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
2019-09-06 10:17:05.488 | INFO     | __main__:main:139 - No local device found, cant start the server

gasparka, 2019-09-07:
Hi, looks like there is no connection to the LimeSDR. 
Just to double-check - have you verified that the unit is functional?

juzhang74, 2019-09-09:
It's my mistake. The program has been run properly!

closed 2019-09-09 by juzhang74

jaanusnuutre, 2019-09-09:
Great!