gasparka/spectrogram #19: [Question] - Using without GQRX - Setting up the client

jjmckee17, 2019-09-26:
First of all, let me say that this project is very well done. I've tried many different implementations of LimeSDR-Mini with Raspberry Pi from across the internet and they were all so painful to set up, taking hours at best. I was up and running with this project in minutes!

I'm trying to set up a logging device that will capture and store a few seconds of FFT data - no need to visualize in real-time. It seems the --server_only option is perfect for this use case. I imagine running the spectrogram server on the pi and then running a python script on the same pi that will connect to the local server, stream in the FFT frames and dump the data to a csv file.

I'm following the usage_demo.ipynb but I'm having trouble connecting to the SoapySDR server. I'm wondering if I should be running my data capture/logging script from inside the docker container? Or if I should manually build/install the SoapySDR/SoapyRemote libraries on the raspberry pi? When I install SoapySDR manually (outside of docker container) I get this error:

`
Traceback (most recent call last):
  File "src/SoapySDRLogger/SoapySDRLogger.py", line 9, in <module>
    sdr_device = SoapySDR.Device({'driver': 'remote', 'cacheCalibrations': '1'})
  File "/usr/local/lib/python2.7/dist-packages/SoapySDR.py", line 1957, in __new__
    return cls.make(*args, **kwargs)
RuntimeError: SoapySDR::Device::make() no match
`

Seems like I'm having trouble connecting through the docker container boundary?

gasparka, 2019-09-26:
First test that the server is actually running:
```bash
user@user-desktop ~> docker ps
CONTAINER ID        IMAGE                      COMMAND                  CREATED             STATUS              PORTS               NAMES
49f8530ac106        gasparka/spectrogram:arm   "/docker_entrypoint.…"   2 minutes ago       Up 2 minutes                            spectrogram_server

```

Next, you need Soapy and SoapyRemote installed (or you can work in Docker, where everything is installed!):
```bash
sudo apt-get install soapysdr-tools soapysdr-module-remote
```

Then you can test for connection:
```bash
user@user-desktop ~> SoapySDRUtil --probe="driver=remote"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device driver=remote
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C45B2BFA379'
[WARNING] This version of LimeSuite only works for the 'spectrogram' application!
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] Selected TX path: Band 2
[INFO] Selected RX path: LNAW
[INFO] LMS7002M calibration values caching Disable

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=0xb2bfa379
  firmwareVersion=5
  gatewareVersion=1.0
  hardwareVersion=2
  protocolVersion=1
  remote:version=0.5.1-unknown

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
  Registers: BBIC
  GPIOs: MAIN
...
```

Also, you can view the server logs:
```bash
user@user-desktop ~> docker logs spectrogram_server
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
######################################################
## Soapy Server -- Use any Soapy SDR remotely
######################################################

Server version: 0.5.1-unknown
Server UUID: 8ab87ab5-ffbe-15c7-8567-0001007f0101
Launching the server... tcp://[::]:55132
Server bound to [::]:55132
Launching discovery server... 
Connecting to DNS-SD daemon... 
[INFO] Avahi version:  avahi 0.7
[INFO] Avahi hostname: user-desktop
[INFO] Avahi domain:   local
[INFO] Avahi FQDN:     user-desktop.local
[INFO] avahi_entry_group_add_service(user-desktop._soapy._tcp)
Press Ctrl+C to stop the server
SoapyServerListener::accept([::ffff:192.168.1.175]:56028)
SoapyServerListener::accept([::ffff:192.168.1.175]:56030)
SoapyServerListener::accept([::ffff:192.168.1.175]:56032)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.175]:56034)
SoapyServerListener::accept([::ffff:192.168.1.175]:56036)
SoapyServerListener::close()
SoapyServerListener::close()
SoapyServerListener::close()

```

Hope this helps!

jjmckee17, 2019-09-26:
Thanks for the quick response! I tried installing soapysdr-tools and soapysdr-module-remote on the pi (not in docker) but the SoapySDRUtil probe command is not finding anything. Here is my terminal output:

`
pi@raspberrypi:~ $ spectrogram --server_only
WARNING:main:Cool your LimeSDR or risk damage!
INFO:main:Pulling latest docker image...
arm: Pulling from gasparka/spectrogram
Digest: sha256:7fbd3ebafb471d30288cd4985ff7b37cbd02fc3ac46f3440a627a25e4e040f95
Status: Image is up to date for gasparka/spectrogram:arm
docker.io/gasparka/spectrogram:arm
INFO:main:Probing for LimeSDR-Mini devices...
INFO:main:remote_available=False, local_available=True, local_fpga_ok=True
INFO:main:Starting server daemon...
d59b51976229ba72f53e78230960e966db0dbbd126d11a57535c2c822768cd76
INFO:main:Server is started!
pi@raspberrypi:~ $ docker ps
CONTAINER ID        IMAGE                      COMMAND                  CREATED             STATUS              PORTS               NAMES
d59b51976229        gasparka/spectrogram:arm   "/docker_entrypoint.…"   29 seconds ago      Up 25 seconds                           spectrogram_server
pi@raspberrypi:~ $ SoapySDRUtil --probe="driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=remote
Error probing device: SoapySDR::Device::make() no match
`

But if I run it from the docker image, it does find the device:
`
pi@raspberrypi:~ $ docker exec -it spectrogram_server /bin/bash
root@raspberrypi:/# SoapySDRUtil --probe="driver=remote"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=remote
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D4257E106E7DA'
[WARNING] This version of LimeSuite only works for the 'spectrogram' application!
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] Selected TX path: Band 2
[INFO] Selected RX path: LNAW
[INFO] LMS7002M calibration values caching Disable

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeSDR-Mini
  boardSerialNumber=0xe106e7da
  firmwareVersion=5
  gatewareVersion=1.0
  hardwareVersion=2
  protocolVersion=1
  remote:version=0.5.2-unknown

...
`

Would you expect the first case to work? Maybe I should create a new docker image, using spectrogram:arm as a base image, and add python3 plus my custom script to collect the data? Or would there be an easier way that you can think o, that wouldn't require me to build a docker image? 

Thanks again for your help!

jjmckee17, 2019-09-26:
I tried the above instructions again, this time on a fresh raspberry pi image. And now `SoapySDRUtil --probe="driver=remote"` IS able to detect the LimeSDR-Mini device, outside of docker! I had previously cloned and manually built the SoapySDR repository - seems apt-get install works better.

Now I need to figure out how to install the python bindings for SoapySDR - ideally without cloning and building the repo. Not sure if it's available through a pip3 install or apt-get install?

gasparka, 2019-09-26:
So this is one situation where pip package would be convenient, which we are discussing at  https://github.com/pothosware/SoapySDR/issues/240.

Currently you should do either:
```
#python language bindings
sudo apt-get install python-soapysdr python-numpy

#python3 language bindings
sudo apt-get install python3-soapysdr python3-numpy
```

jjmckee17, 2019-09-26:
Installing python3-soapysdr has done the trick! Thank you so much for your help.

closed 2019-09-26 by jjmckee17

gasparka, 2019-09-26:
Great!