gasparka/spectrogram #20: [Question] Support for Raspberry Pi Zero W?

jjmckee17, 2019-10-02:
Just wondering if anyone has had any success running this on a Raspberry Pi Zero W?
There is an extra step required to set up docker [here](https://markmcgookin.com/2019/08/04/how-to-install-docker-on-a-raspberry-pi-zero-w-running-raspbian-buster/).
Once docker is properly installed, the spectrogram app runs, but it doesn't detect the LimeSDR-Mini device.
```
2019-10-02 12:46:48.843 | INFO     | __main__:probe_devices:83 - Probing for LimeSDR-Mini devices...
2019-10-02 12:47:36.379 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
2019-10-02 12:47:36.404 | INFO     | __main__:main:139 - No local device found, cant start the server
```

gasparka, 2019-10-02:
What is the output of ```SoapySDRUtil --probe```?

jjmckee17, 2019-10-02:
```
pi@raspberrypi:~ $ SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
Error probing device: device_id missing.
```

gasparka, 2019-10-02:
Strange stuff. Lets try inside the docker image:
 ```bash
docker run -i --privileged --net=host gasparka/spectrogram:arm SoapySDRUtil --probe="driver=lime"
```

jjmckee17, 2019-10-02:
Wondering if I'm having issue with docker on the Raspberry Pi Zero W device. This command exits immediately with a return code of 139 (segfault). It doesn't output anything to the terminal and `docker logs` also outputs nothing.

Running the generic `docker run -it ubuntu bash` also does the same, but normally this should run the container and give me a terminal inside.

gasparka, 2019-10-02:
Ok that is a surprise to me. There is not even Ubuntu docker image for arm32v6, thus porting looks not trivial. I feel this is probably not worth the effort, but if you like you can try to ditch docker and just install everything manually (but i estimate you end up using alot of time). You can look at the https://github.com/gasparka/spectrogram/blob/master/Dockerfile.arm on what needs to be installed.
Of course, i am able to help if you get stuck.

Also maybe take a look at some modern compute modules, like Jetson Nano - i have already verified that everything works there :)

jjmckee17, 2019-10-02:
Thanks again for your great support. I agree that it's probably not worth the effort to port to PiZero. I could go through the effort of installing everything manually - likely just to hit the next dependency/performance roadblock with another section of my project! Better to stick with the more capable and mature Pi 3.

closed 2019-10-02 by jjmckee17