gasparka/spectrogram #22: What's the problem when it is at 140MHz?

juzhang74, 2020-03-10:
Hi, 
 I meet a strange problem with this program.
 When it running at 2.4GHz, all seems OK.
 If I set the central freq at 140MHz, with GNURadio and the official FPGA, all things seems OK, but at the same central freq, spectrogram's outputs has nothing except noise floor.
Could anyone tell me why? Any advice will be appreciated.

gasparka, 2020-03-10:
Hi
You can press Control + j to open SDR configuration sidebar. Antenna selection should be already at "Auto", but maybe you need to increase the gain?

juzhang74, 2020-03-11:
Yes-ish!!!
I can see signals at 140MHz with LNA gain set to 20.
Thanks a lot!

closed 2020-03-11 by juzhang74

gasparka, 2020-03-11:
LNA should be by default set to 30. Did you mean VGA, or maybe we are dealing with some bug here?

juzhang74, 2020-03-13:
We use your usage_demo.ipynb demo. And in this demo, Default gains are set to "gains = {LNA: 0, TIA: 0, PGA: -12}". That's the problem. When I set LNA gain to 20, all seems Ok!

juzhang74, 2020-03-13:
Thank you very much!