gasparka/spectrogram #23: Req: updated guide to install on ubuntu 20.04 LTS amd64 Desktop

tdbone1, 2020-10-02:
updated guide please as i am receiving errors when i run ./local/bin/spectrogram.
if i try and install spectrogram with 'pip install spectrogram' it does not complete so i have to use 'pip3 install spectrogram' so i hope this is ok on an x86_64 and not the rpi3.

after that is finished installing i then 'cd .local/bin' and perform ./spectrogram but get errors so i am then forced to use './spectrogram --server_only' which does complete and says server started.
i then use gqrx from the snap-store with these arguments "soapy=0,driver=remote,remote:driver=lime" and i configure gqrx with 80msps and 80mhz bandwidth, demodulate = disabled, 512fft, 60fps waterfall speed, antenna = auto and i set frequency to 2440mhz.
i dont believe im getting expected behavior can someone please tell me correct way to do this or if i did do it correctly?
thanks and i love this great work you do.
awesome program and thanks for your hard work gasparka.

gasparka, 2020-10-04:
You cant use regular GQRX to connect to the server. Always must use the modified one which is inside the docker image. I see that there are problems running the GQRX in Ubuntu 20. Currently i would suggest falling back to Ubuntu 18. Maybe i will get some time to investigate..but cant make any promises!

tdbone1, 2020-10-14:
Dell Laptop running ubuntu amd64 desktop
bgbone1@3540:~$ uname -a
Linux 3540 5.4.0-51-generic #56~18.04.1-Ubuntu SMP Tue Oct 6 09:47:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
bgbone1@3540:~$ lsb_release -r
Release:	18.04

raspberry pi3b running raspberry pi os lite
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ lsb_release -r
Release:	10
i sure the rpi3b as headless (no mouse keyboard or hdmi)

i use my phone (LG V50 on Android 10) as a wifi hotspot to connect the laptop to the rpi3b over wifi with ssh

I plug the limesdr-mini into the pi3b and do this command:

pi@raspberrypi:~ $ .local/bin/spectrogram --server_only
2020-10-14 06:15:42.207 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
2020-10-14 06:15:44.480 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
arm: Pulling from gasparka/spectrogram
Digest: sha256:7fbd3ebafb471d30288cd4985ff7b37cbd02fc3ac46f3440a627a25e4e040f95
Status: Image is up to date for gasparka/spectrogram:arm
docker.io/gasparka/spectrogram:arm
2020-10-14 06:15:48.300 | INFO     | __main__:probe_devices:83 - Probing for LimeSDR-Mini devices...
2020-10-14 06:15:59.442 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=True, local_fpga_ok=True
2020-10-14 06:15:59.443 | INFO     | __main__:docker_start_server_daemon:58 - Starting server daemon...
6bc50275898f8cbb8b8572b2fb3e89c1213694b497054880d851dbb43202e511
2020-10-14 06:16:03.558 | INFO     | __main__:main:144 - Server is started!
pi@raspberrypi:~ $ 

i then issue this command on my Dell Laptop:

bgbone1@3540:~$ .local/bin/spectrogram
2020-10-14 06:17:27.013 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
2020-10-14 06:17:27.345 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
Digest: sha256:bc75b4cea042f098e926cc7e4b100c2e8cbbbea5eaeb176a70bb1ac419bb8dc2
Status: Image is up to date for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2020-10-14 06:17:29.731 | INFO     | __main__:probe_devices:83 - Probing for LimeSDR-Mini devices...
2020-10-14 06:17:31.828 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
2020-10-14 06:17:31.829 | INFO     | __main__:main:170 - Done!
bgbone1@3540:~$ 

????

ok so i now do:

bgbone1@3540:~$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  addr = 24607:1027
  driver = remote
  label = LimeSDR Mini [USB 2.0] 1D3BD0C4F0CDFF
  media = USB 2.0
  module = FT601
  name = LimeSDR Mini
  remote = tcp://192.168.43.47:55132
  remote:driver = lime
  serial = 1D3BD0C4F0CDFF

bgbone1@3540:~$ 

why is the laptop not starting gqrx docker image?

gasparka, 2020-10-14:
Lets focus on the solo laptop case first. I see that it is running Ubuntu 18, but still there is the X display issue as in Ubuntu 20.
I try to reproduce this during weekend.

tdbone1, 2020-10-14:
ok i did this on the laptop only with no rpi3b and here is the results:

bgbone1@3540:~$ uname -a
Linux 3540 5.4.0-51-generic #56~18.04.1-Ubuntu SMP Tue Oct 6 09:47:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
bgbone1@3540:~$ lsb_release -r
Release:	18.04
bgbone1@3540:~$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  addr = 24607:1027
  driver = lime
  label = LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF
  media = USB 3.0
  module = FT601
  name = LimeSDR Mini
  serial = 1D3BD0C4F0CDFF

bgbone1@3540:~$ LimeUtil --find
  * [LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D3BD0C4F0CDFF]

bgbone1@3540:~$ .local/bin/spectrogram --fpga_restore
2020-10-14 12:34:57.950 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
2020-10-14 12:34:58.016 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
Digest: sha256:bc75b4cea042f098e926cc7e4b100c2e8cbbbea5eaeb176a70bb1ac419bb8dc2
Status: Image is up to date for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2020-10-14 12:34:59.366 | INFO     | __main__:probe_devices:83 - Probing for LimeSDR-Mini devices...
2020-10-14 12:35:02.191 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=True, local_fpga_ok=True
2020-10-14 12:35:02.191 | INFO     | __main__:docker_restore_fpga:76 - Restoring FPGA, takes ~20 sec...
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
Connected to [LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF]
--2020-10-14 17:35:02--  http://downloads.myriadrf.org/project/limesuite/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd
Resolving downloads.myriadrf.org (downloads.myriadrf.org)... 134.209.28.241, 2a03:b0c0:1:e0::3f9:a001
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|134.209.28.241|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.myriadrf.org/project/limesuite/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd [following]
--2020-10-14 17:35:03--  https://downloads.myriadrf.org/project/limesuite/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd
Connecting to downloads.myriadrf.org (downloads.myriadrf.org)|134.209.28.241|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 577536 (564K) [application/octet-stream]
Saving to: '/root/.local/share/LimeSuite/images/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd'

     0K .......... .......... .......... .......... ..........  8%  168K 3s
    50K .......... .......... .......... .......... .......... 17%  352K 2s
   100K .......... .......... .......... .......... .......... 26%  422K 2s
   150K .......... .......... .......... .......... .......... 35%  486K 1s
   200K .......... .......... .......... .......... .......... 44%  507K 1s
   250K .......... .......... .......... .......... .......... 53%  409K 1s
   300K .......... .......... .......... .......... .......... 62%  373K 1s
   350K .......... .......... .......... .......... .......... 70%  521K 0s
   400K .......... .......... .......... .......... .......... 79%  678K 0s
   450K .......... .......... .......... .......... .......... 88%  372K 0s
   500K .......... .......... .......... .......... .......... 97%  730K 0s
   550K .......... ....                                       100%  320K=1.4s

2020-10-14 17:35:05 (394 KB/s) - '/root/.local/share/LimeSuite/images/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd' saved [577536/577536]

[100%] 303104/303104 Bytes programming: completed (/root/.local/share/LimeSuite/images/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd)
Programming update complete!
bgbone1@3540:~$ .local/bin/spectrogram
2020-10-14 12:35:37.155 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
2020-10-14 12:35:37.222 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
[... 9 lines cut ...]
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
Connected to [LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF]
This version of LimeSuite only works for the 'spectrogram' application!
Reference clock 40.00 MHz
[100%] 303104/303104 Bytes programming: completed
2020-10-14 12:36:00.965 | INFO     | __main__:docker_start_server_daemon:58 - Starting server daemon...
08b033b631bccec4f1e3398b754e70eeb39a83bc38fcddbfd1cc2cdc2f304783
2020-10-14 12:36:01.360 | INFO     | __main__:docker_start_gqrx:48 - Starting gqrx...
2020-10-14 12:36:01.361 | WARNING  | __main__:warn_temperature:16 - Cool your LimeSDR or risk damage!
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
QStandardPaths: error creating runtime directory /tmp/runtime-root: No such file or directory
dbus[1]: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
2020-10-14 12:36:02.244 | INFO     | __main__:main:167 - Killing local server...
spectrogram_server
2020-10-14 12:36:02.539 | INFO     | __main__:main:170 - Done!
bgbone1@3540:~$ 

why cant i get './spectrogram' to work on ubuntu 18.04.5 amd64 Desktop?

gasparka, 2020-10-21:
New version should work on Ubuntu 20, would be great if you can test it and report back!

tdbone1, 2020-10-21:
> New version should work on Ubuntu 20, would be great if you can test it and report back!

i just did a completely fresh install of ubuntu 20.04 amd64 desktop (Daily) on both my laptops.
is there any certain thing you want me to try before i run the 'pip3 install spectrogram' command?
note:
on my i7 laptop i am using the nvidia meta binary package and on my i3 laptop its just using the nouveau driver.
everything right now is updated and fresh default install on both systems.

gasparka, 2020-10-21:
No need for anything special. If you encounter some troubles we can write them up in the README, so people after you have easier time!

tdbone1, 2020-10-21:
ok so here is what i did so far on clean ubuntu 20.04 amd64 desktop install and fully updated.

pip3 install spectrogram
sudo apt install python3-pip
sudo apt install curl
curl -fsSL https://get.docker.com | sh
sudo groupadd docker 
sudo usermod -aG docker $USER 
# Log out and log back in so that your group membership is re-evaluated.

en1gma@M4800:~/.local/bin$ ./spectrogram 
2020-10-21 12:54:56.497 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
2020-10-21 12:54:56.530 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/images/create?fromImage=gasparka%2Fspectrogram&tag=latest: dial unix /var/run/docker.sock: connect: permission denied
2020-10-21 12:54:56.563 | INFO     | __main__:probe_devices:84 - Probing for LimeSDR-Mini devices...
2020-10-21 12:54:56.631 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
non-network local connections being removed from access control list
2020-10-21 12:54:56.634 | INFO     | __main__:main:172 - Done!
en1gma@M4800:~/.local/bin$

tdbone1, 2020-10-21:
here is the I3 Laptop running ubuntu 20.04 amd64 desktop (Daily) fully updated.

pip3 install spectrogram
sudo apt install python3-pip
sudo apt install curl
curl -fsSL https://get.docker.com | sh
sudo groupadd docker
sudo usermod -aG docker $USER
# Log out and log back in so that your group membership is re-evaluated.

bgbone1@L3540:~/.local/bin$ ./spectrogram
2020-10-21 12:55:11.010 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
2020-10-21 12:55:11.073 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/images/create?fromImage=gasparka%2Fspectrogram&tag=latest: dial unix /var/run/docker.sock: connect: permission denied
2020-10-21 12:55:11.131 | INFO     | __main__:probe_devices:84 - Probing for LimeSDR-Mini devices...
2020-10-21 12:55:11.251 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
non-network local connections being removed from access control list
2020-10-21 12:55:11.256 | INFO     | __main__:main:172 - Done!
bgbone1@L3540:~/.local/bin$

gasparka, 2020-10-21:
You need to log out and back in after installing the docker. You may also just restart the system.

tdbone1, 2020-10-21:
> You need to log out and back in after installing the docker. You may also just restart the system.

Restarting the laptop worked.
[https://imgur.com/a/1vETnjw](url)

i havent tried the ',.spectrogram --server_only' or './spectrogram --fpga_restore' yet.
also i havent tried connecting from laptop #2 running './spectrogram' and laptop #1 running './spectrogram --server_only'
going to do some testing but as far as running on ubuntu 20.04 amd64 it looks good with directions above except 'python3-pip' should be ran before 'pip3 install spectrogram' otherwise directions are good.

will report back of how testing goes with 2 laptops connected to cellphone wifi hotspot.
thanks gasparka!

tdbone1, 2020-10-21:
ok more info.
laptop #1 spectrogram --server_only
laptop #2 spectrogram

bgbone1@L3540:~/.local/bin$ ./spectrpgra,
bash: ./spectrpgra,: No such file or directory
bgbone1@L3540:~/.local/bin$ ./spectrpgram
bash: ./spectrpgram: No such file or directory
bgbone1@L3540:~/.local/bin$ ./spectrogram 
2020-10-21 13:36:50.351 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
2020-10-21 13:36:50.696 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
5667fdb72017: Pull complete 
d83811f270d5: Pull complete 
ee671aafb583: Pull complete 
7fc152dfb3a6: Pull complete 
1f66a4a29d15: Pull complete 
c1eb5fda0c10: Pull complete 
e5a55465a2c5: Pull complete 
e726728ac642: Pull complete 
b43e8397930c: Pull complete 
3ef9a0f70809: Pull complete 
1a7e40ea62d9: Pull complete 
4a531422f233: Pull complete 
9f7259efeea8: Pull complete 
02aa2496f529: Pull complete 
9a7497286fcd: Pull complete 
89e27304a215: Pull complete 
Digest: sha256:bc75b4cea042f098e926cc7e4b100c2e8cbbbea5eaeb176a70bb1ac419bb8dc2
Status: Downloaded newer image for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2020-10-21 13:37:56.242 | INFO     | __main__:probe_devices:84 - Probing for LimeSDR-Mini devices...
2020-10-21 13:38:00.780 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=False, local_fpga_ok=False
non-network local connections being removed from access control list
2020-10-21 13:38:00.788 | INFO     | __main__:main:172 - Done!
bgbone1@L3540:~/.local/bin$

tdbone1, 2020-10-21:
this is all the info from my cli on laptop #1 
en1gma@M4800:~$ cd .local/bin
en1gma@M4800:~/.local/bin$ ./spectrogram 
2020-10-21 13:23:38.462 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
2020-10-21 13:23:38.719 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
5667fdb72017: Pull complete 
d83811f270d5: Pull complete 
ee671aafb583: Pull complete 
7fc152dfb3a6: Pull complete 
1f66a4a29d15: Pull complete 
c1eb5fda0c10: Pull complete 
e5a55465a2c5: Pull complete 
e726728ac642: Pull complete 
b43e8397930c: Pull complete 
3ef9a0f70809: Pull complete 
1a7e40ea62d9: Pull complete 
4a531422f233: Pull complete 
9f7259efeea8: Pull complete 
02aa2496f529: Pull complete 
9a7497286fcd: Pull complete 
89e27304a215: Pull complete 
Digest: sha256:bc75b4cea042f098e926cc7e4b100c2e8cbbbea5eaeb176a70bb1ac419bb8dc2
Status: Downloaded newer image for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2020-10-21 13:24:30.895 | INFO     | __main__:probe_devices:84 - Probing for LimeSDR-Mini devices...
2020-10-21 13:24:34.468 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=True, local_fpga_ok=False
2020-10-21 13:24:34.468 | INFO     | __main__:docker_program_fpga:69 - Programming FPGA, takes ~20 sec...
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
Connected to [LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF]
This version of LimeSuite only works for the 'spectrogram' application!
Reference clock 40.00 MHz
[100%] 303104/303104 Bytes programming: completed
2020-10-21 13:24:54.614 | INFO     | __main__:docker_start_server_daemon:59 - Starting server daemon...
43af522c5008783f85e713deefb667d2742efc0791bcb0f091df2f25140017db
2020-10-21 13:24:54.804 | INFO     | __main__:docker_start_gqrx:48 - Starting gqrx...
2020-10-21 13:24:54.805 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
non-network local connections being added to access control list
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
QStandardPaths: error creating runtime directory /tmp/runtime-root: No such file or directory
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF'
[WARNING] This version of LimeSuite only works for the 'spectrogram' application!
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] Selected TX path: Band 2
[INFO] Selected RX path: LNAW
[INFO] LMS7002M calibration values caching Enable
setting samples rates to 80 Msps...
[INFO] Setting oversampling of 2
[INFO] Selected RX path: LNAH
[INFO] Setting oversampling of 2
[INFO] RX LPF configured
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=2048, mtu=2120, window=44040192)
[INFO] Server side stream bound to [::ffff:172.17.0.1]:34177
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 208 KiB
[INFO] Configured sender endpoint: dgram=2072 bytes, 512 elements @ 4 bytes, window=208 KiB
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 208 KiB
[INFO] Configured receiver endpoint: dgram=2072 bytes, 512 elements @ 4 bytes, window=208 KiB
[INFO] Selected RX path: LNAH
[INFO] Selected RX path: LNAH
[INFO] RX LPF configured
[INFO] Selected RX path: LNAW
[INFO] Selected RX path: LNAH
[... 101 lines cut ...]
[100%] 303104/303104 Bytes programming: completed (/root/.local/share/LimeSuite/images/18.10/LimeSDR-Mini_HW_1.2_r1.28.rpd)
Programming update complete!
en1gma@M4800:~/.local/bin$ ./spectrogram --server_only
2020-10-21 13:35:29.151 | WARNING  | __main__:warn_temperature:15 - Cool your LimeSDR or risk damage!
2020-10-21 13:35:29.186 | INFO     | __main__:docker_pull:20 - Pulling latest docker image...
Using default tag: latest
latest: Pulling from gasparka/spectrogram
Digest: sha256:bc75b4cea042f098e926cc7e4b100c2e8cbbbea5eaeb176a70bb1ac419bb8dc2
Status: Image is up to date for gasparka/spectrogram:latest
docker.io/gasparka/spectrogram:latest
2020-10-21 13:35:30.820 | INFO     | __main__:probe_devices:84 - Probing for LimeSDR-Mini devices...
2020-10-21 13:35:32.683 | INFO     | __main__:probe_devices:106 - remote_available=False, local_available=True, local_fpga_ok=False
2020-10-21 13:35:32.683 | INFO     | __main__:docker_program_fpga:69 - Programming FPGA, takes ~20 sec...
 * Starting system message bus dbus
   ...done.
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon
   ...done.
Connected to [LimeSDR Mini [USB 3.0] 1D3BD0C4F0CDFF]
This version of LimeSuite only works for the 'spectrogram' application!
Reference clock 40.00 MHz
[100%] 303104/303104 Bytes programming: completed
2020-10-21 13:35:52.823 | INFO     | __main__:docker_start_server_daemon:59 - Starting server daemon...
5030eda9cf3fb495826b60346c52041dc7566e0ec583b14fecb7c6119c257cca
2020-10-21 13:35:53.020 | INFO     | __main__:main:145 - Server is started!
en1gma@M4800:~/.local/bin$

tdbone1, 2020-10-21:
one more thing also gasparka.
i think spectrogram --fpga_restore = 1.28 but i think i normally am on 1.30 gateware
no big deal because i can use limesuitegui or limeutil to auto update but i figure i might tell you since you are here

tdbone1, 2020-10-21:
wait!

ok i put limesdr-mini on laptop #2 and did ./spectrogram --server_only and it says Server is started!

on laptop #1 i did ./spectrogram and its working.

not sure why it didnt work the other way reveresed.
going to try again

tdbone1, 2020-10-21:
well....
im not sure what is going on because i installed ubuntu 20.04 amd64 deskop (daily image) to both laptops (exactly the same) except for laptop #1 has an I7 and an nvidia dedicated gpu and the laptop #2 has only the I3 with built-in intel gpu.

i dont think the reason why its not connecting from laptop #2 to laptop #1 but it works when i do it from laptop #1 to laptop #2 
i will check to make sure the rpi3b can run 'spectrogram --server_only' and i will try and connect both laptops to the rpi3b and report back.
give me a bit of time

tdbone1, 2020-10-21:
@gasparka 
ok i got to do quite a bit of testing.
on laptop #1 with the I7-4770 and the dedicated nvidia gpu i changed the nvidia X-Server to performace profile to make it use my onboard intel gpu instead of the nvidia gpu.
that got rid of 2 small errors.

RPI3B = 2020-08-20-raspios-buster-armhf
I7-4770 = ubuntu 20.04.1 amd64 desktop (minimal install and updated
I3-4005 = ubuntu 20.04.1 amd64 desktop (minimal install and updated)
these 3 devices are connected to my phones wifi hotspot.

everything is working fine.
'./spectrogram' seems to work more stable when it is ran locally from same machine (not connected to a remote device)
when connecting to a remote machine it is a little glitchy but it will connect.
if any device has been running './spectrogram' locally with no remote connection it is a good idea to reset/shutdown the device before starting './spectrogram --server_only' as it leaves gqrx open (i think).
if the server is started successfully with './spectrogram --server_only' it make take quite a few tries to have a succeful connection with './spectrogram' on the remote device.
im not sure if this is just a time thing or if it just is having problems "Probing" the server. if you do it enough it will connect.

i think everything is working gasparka.
!GREAT WORK!
awesome program!
if i wanted to use one of my two laptops (both have usb 3.0 ports) as a server i think i could keep my lid closed and it can power the limesdr-mini for a long time i bet.
the laptops have alot better cpu then the rpi3b.
for the laptops is there an easy way to adjust the fixed 1MB/s bandwidth? i think even the rpi3b can do 2MB/s
i did alot of testing with the rpi3b with iperf3 (i was getting 20-30mbps but it was not running ',/spectrogram --server_only'.

anyhow this is a really awesome project and you do great work.
i know i already said that but its pretty cool.

if you need any logs or info just let me know and i will get you what i can.

tdbone1, 2020-10-21:
i would make an updated guide but i think your guide is working as of this date.
ubuntu 20.04 amd64 Desktop and rpi3b 2020-08-20-raspios-buster-armhf is also working.
not sure what to say except if you switch from running './spectrogram' locally with no network connections make sure to restart your device if you switch to './spectrogram --server_only'
i have a couple errors about libgl.so or something but that switched when i set my gpu to intel.
the errors didnt stop ./spectrogram from running but it seemed like getting a remote computer to connect can take a bit of "time" when trying to connect to server remotely.
not sure but restarting the devices when changing modes might have helped or fixed that.

if you want me to make a guide to install i can but i was looking at your guide and i think its still good now.

gasparka, 2020-10-22:
I have added some details to the install instructions. Thanks for testing. I am glad my work is useful for someone :)

closed 2020-10-22 by gasparka