gasparka/spectrogram #25: Ubuntu 18.04 Desktop amd64 64bit and 'pip install spectrogram' or 'pip3 install spectrogram'

tdbone1, 2020-10-19:
I am requesting help.

can anyone please do a fresh install of ubuntu on a 64bit pc and tell me the process of running limesdr-mini at 80mhz?

i can do this easy on my rpi3b with raspberrypios 2020-08-20.

i have been messing around with this far to long.
ubuntu 18.04 is said to work in the instructions.
it does not.

gasparka, 2020-10-20:
I will test today. Sorry i kept you waiting, some busy times at work.

gasparka, 2020-10-20:
I verified that there is an issue on Ubuntu. Still works on my Manjaro i3 edition computer, but this is not a fresh install.
I am hoping to fix this during this week, will keep you posted on progress.

tdbone1, 2020-10-20:
@gasparka its quite alright.
you are not keeping me waiting i was thinking another person might help as i know you are busy doing more important work.
i have two laptops. one laptop = i3-4005u and my other laptop = i7-4770. both are 4th generation intel and of course i have a rpi3b.

cant wait for new release of 'spectrogram' for updated ubuntu amd64.
ubuntu 20.04 LTS?
thanks and by

yea i love what you did for the rpi3 as it just works flawless for the regular './spectrogram' with no options. gqrx opens up and just starts working. very nice program.

cant wait to see how it really works connected to something more powerful then a rpi3b.
these x86_64 cpus with usb3 and 5ghz wifi should make an awesome './spectrogram --server_only' on one of the laptops and './spectrogram' on the other laptop.

gasparka, 2020-10-21:
I believe i have fixed the problem. You need to update your "spectrogram" script from pip:
```pip3 install spectrogram --upgrade```
Version 1.0.9 should be installed.

Let me know how it goes!

gasparka, 2020-10-21:
Actually the new version is not pushed yet, it is waiting at Travis CI. Dont know what is taking so long.

tdbone1, 2020-10-21:
> I believe i have fixed the problem. You need to update your "spectrogram" script from pip:
> `pip3 install spectrogram --upgrade`
> Version 1.0.9 should be installed.
> 
> Let me know how it goes!

how do i updated the "spectrogram" script?
do i just change all instances of "pip" to "pip3"?

gasparka, 2020-10-21:
To update run:
```pip3 install spectrogram --upgrade```

gasparka, 2020-10-22:
Solved!

closed 2020-10-22 by gasparka