gasparka/spectrogram #27: Documentation on what the new FPGA code actually streams back to over the USB connection.

wgaylord, 2021-05-21:
I am looking into trying to sort of modify STRF (https://github.com/cbassa/strf) to work with this and was wondering what actually is streamed back, as normally SoappySDR streams I/Q samples not FFT bins.

gasparka, 2021-05-21:
https://github.com/gasparka/spectrogram/blob/master/doc/usage_demo.ipynb
Should provide a good example on how to deal with the stream. 
In FPGA 512 point FFT is calculated, then 256 of these are averaged and result sent over USB.
FFT points and averaging amount could be changed if need arises.
Happy to help if i can :)

wgaylord, 2021-05-21:
Okay, that sounds useful. :)

It sound be simple enough to make it into something that strf can understand. 

Also what sample rate is it running at normally? As for STRF you generally want about 100hz wide bins meaning that the total bandwidth I would see with a 512 point FFT is 5 Khz-ish of spectrum provided I set the sample rate to get 100hz wide bins.

gasparka, 2021-06-03:
Sorry i was away. Anyhow. Sample rate is 80 MHz. I believe it also worked with 40 MHz I dont know if it works with lower sample rate, because there might be some FPGA sampling issues - Lime SDR chip has a nasty - i would call it a bug - where optimal sampling point depends on the sample rate. Idea of this project mas to process big bandwidth, so lower sample rate is not tested at all.